JonMuller/gerbers #1: Keep Tenax Licence

toniz4, 2023-03-03:
Hey! Pretty nice project.

If possible, can you follow the copying section of the tenax licence? You can get it here:

https://github.com/toniz4/tenax/blob/master/LICENSE

You just need to retain the notices and state what you changed, there's no need to provide all the source files necessairly.

JonMuller, 2023-03-03:
I'd be happy to comply. I'm not really sure exactly what that means. I can copy that license into the Tenax folder. If I'm being honest, I don't know that I fully understand how or where to document what I changed. I'm more than happy to do that though. If you have any tips or examples, I would appreciate it. I'm going to do some googling now to see what else I can do.

toniz4, 2023-03-04:
I think that would be enough. Thanks!

And btw, if you already built one, can you post some pics? I can add it to the main repo and link to your project, if you don't mind.

JonMuller, 2023-03-04:
Will do. I did build one and it's working without issue so far! I meant to go back and update this with a tested/verified flag do some sort too.

JonMuller, 2023-03-11:
Ok, took me a bit, but I finally got the pics added. I also added in the files necessary to get things up and running on ZMK. If that works for you, I'll go ahead and close this issue.

Thanks!

toniz4, 2023-03-12:
Perfect

closed 2023-03-12 by toniz4