JonMuller/gerbers #2: Questions regarding battery life & 3D print case

j1n6, 2023-03-25:
Hi there, thanks for sharing the design. They look absolutely interesting. I'm planning to build one for myself. 
I have a couple of questions, what is the battery life for the Corne build? Would you able to share the baseplate 3D print file? 

Thanks!

JonMuller, 2023-03-25:
Hi! Battery life is really hard to pin down. It really depends on the size of battery that you use. I'm getting at least a couple of weeks with the joycon batteries. That's got a lot to due with the displays being pretty power efficient. If using the OLED displays, it's probably a bit worse. I have some 800 mAh batteries on the one with OLEDs and it's been going for about 2 weeks now. Obviously no displays will get you the best battery life.

I'm working on the case files right now. They are still a work in progress, so I'd put a bit of a caution on those. I'll try to upload them soon.

j1n6, 2023-03-25:
That's pretty good battery life! 
I will order the pcbs, will wait for the 3D files. Thank you for sharing your design, love it!

JonMuller, 2023-03-26:
Case files have been uploaded. They are still a bit of a work in progress though. FYI.

tangbonze, 2023-03-27:
Thank you for share case design. Any chance also share stl for 6col pcb? Thanks.

JonMuller, 2023-03-28:
Unfortunately, I do not have a 6 col version at this time. It won't be too difficult to make one, I just don't have time right now. I might be able to get to it in a few weeks though.

tangbonze, 2023-03-30:
Can I use a corne case instead?

JonMuller, 2023-04-09:
Sorry, missed this. I don't think you can use a corne case. I moved the mount holes and I'm not sure the outline is the same. It would be iffy at best.

closed 2023-06-03 by j1n6