JonMuller/gerbers #4: Questions about the location of the diode

lu-kan, 2023-04-09:
It surprised me so much that I managed to make a working one just from the imgur pics, without a real build guide. Thank you so much!
Just want to confirm one thing: for the boards to work, the diodes MUST be soldered to the back of the left shield and the top of the right shield, right?

I think a tilted case will make room for even bigger batteries, so I am trying to learn CAD to make a stl for the 6-col case.

JonMuller, 2023-04-09:
If you're referring to the corne spaced XIAO, then no. It doesn't matter. I was having the diodes done at the factory, so it's much cheaper to do them all on one side. All that matters is that they are facing the right direction as far as polarity is concerned. You can place them all on the front or all on the back, whichever suits you better. You want the cathode to face the circled hole ideally, but it's configurable in whatever firmware you use. Most important is that they all are soldered the same orientation. 

Hope that makes sense!

JonMuller, 2023-04-09:
Also, thanks for pointing out how helpful those pics were. I added a link.

JonMuller, 2023-04-14:
Question has been answered. Closing this issue.

closed 2023-04-14 by JonMuller