JonMuller/gerbers #6: Kicad files?

vloth, 2023-04-25:
Thanks for sharing the project, these are some fantastic looking keebs!
I'm considering building some corne xiaos and was thinking in creating a FR4 sandwich case for 6 columns. I think I might know how to do it from the kicad. Are you planning to add the kicad files in future?
Thanks again!

JonMuller, 2023-04-27:
Thanks!
I don't have any plans at this time to sort out the kicad, but I did include the SVG files for what the plate would look like. You should be able to open a new project in kicad and in the PCB editor, File->Import->Graphics, then select the version you want, 5 or 6 col, then make sure you're on the Edge Cuts layer and you should be good. You'll have to clean it up a little more because I left in extra information you can you to reference where the display and controller go. But that shouldn't be too difficult I don't think.

vloth, 2023-04-27:
Thanks for the quick reply, I'll try the suggested method. 👍

closed 2023-04-27 by vloth