JonMuller/gerbers #8: 6 column Xiao BLE Corne Firmware

j1n6, 2023-06-03:
I managed to solder a set together, this looks beautiful! Thank you for sharing this. 
I'm now progressing to firmware... two questions:

1) any specific Changes I need to make for the firmware to support 6 columns?
2) I seem to have issues with the right half: row 0 and row 1 creates strange double keys.
     for example, if I type 'j', it output 'uj'; if I type 'u'. it also output 'uj'
     I double checked soldering, it seems to be ok. 

Any thoughts? 

Thank you.

JonMuller, 2023-06-05:
Yup. Sorry. I should probably include some more info for the 6 col version. You did it right though.

manna-harbour, 2023-08-07:
Here's a fix for the missing col-offset: https://github.com/JonMuller/gerbers/pull/12.

To select the 6-column transform, add this to your keymap: https://github.com/manna-harbour/miryoku_zmk/blob/b2795e263bb8ec80e2233135206befaa59776248/config/chipper.keymap#L8-L12.