JonMuller/gerbers #9: LCSC part numbers?

JonnyHaystack, 2023-06-24:
Hi,

Love your corne choc xiao design! Is there any chance you could give some example LCSC part numbers for the diodes and molex connectors? Even if the particular part is not always in stock, it's easier to find an equivalent part when given a reference part to go off.

MuhametSmaili, 2023-07-04:
@JonnyHaystack I am checking to order this pcb too in the coming weeks and I was searching and checking the same things. 

@JonMuller  as stated on the ReadeMe file he ordered the PCB via assembly hence the diodes and Molex came soldered into place. 

Here are the parts that I could find (@JonMuller  can confirm)
- [Molex](https://www.digikey.fi/en/products/detail/molex/0781710002/2424924) -> Pico EZmate 1x02-MP_P1.20mm 
- [Diodes I think normal SMD-123 diodes ](https://www.aliexpress.com/item/1005004188010012.html?algo_exp_id=dcf3b60b-d255-499d-b497-787991bbef9f-0)

Not 100% sure, hope @JonMuller  can confirm. These are the parts I will go If I decide to print in the near future.

Things I am checking now are: 

- Printing a 6-column case for the board. (will try to modify the Jons' case to support 6 columns, but I do not own a 3d printer nor designed something myself in the past, so it will be a bit hard)
-  I will try to make the PCB not breakable to 5 columns because I think I can break on my bag I am sure that I will want 6 columns for now. I do not think I can adjust and learn typing on the 5-column layout. 

Hope I could help.

JonMuller, 2023-07-05:
Hi all!
I'm sorry I didn't see this sooner!
@MuhametSmaili is correct on the parts. But to clarify, here's the LCSC #'s:
C505023 - Pico EZmate
C2972761 - diodes

I should probably update the BOM. Can be tricky when stock is out though.

Hope that helps!

JonnyHaystack, 2023-07-05:
Thanks both! I actually noticed after creating this issue that you do have the LCSC part numbers in the BOM for your tenax xiao PCB in this repo, just not for the corne choc xiao. I think both parts were well stocked also.

PR #11 (2023-08-08): Add LCSC part numbers to corne-choc-xiao BOM
Resolves #9
closed 2023-08-08 by JonMuller