JonMuller/gerbers #10: Case for 6 column version?

Kaesono, 2023-07-10:
Is there a case somewhere for the 6 column version? Am I missing something? Or how would I be able to create my own 6 column case if so

JonMuller, 2023-07-10:
You aren't missing anything. I only made a case for the 5 column version because that's what I use. I did include an SVG file for the  6 column version though. You can use that as a starter and follow the instructions here (https://kbd.news/Keyboard-Case-Design-1764.html) to create something similar. It's a bit of a learning curve, but a very useful skill in this hobby.

Kaesono, 2023-07-10:
Is there a way that you or I could change your 5 column version of the case by extending it to have the sixth column?

JonMuller, 2023-07-10:
You might be able to, but it wouldn't be that pretty I don't think. The supports for the way you'd put a 6 column in there would need to be added. 

I personally think it would be easier to just make one from scratch with the SVG file. I'm not at all opposed to doing that, but I just haven't had the time. You could try loading the current 5 column file into tinkercad online and seeing if you could split it towards the outer column, then moving it over and filling in the gap. That's pretty iffy though if you ask me. Especially if you're paying someone else to print it for you. 

That' the other time issue for me. I'd like to actually print it and make sure it works before sharing. However, if I win the lottery tonight, I promise I'll have something for you by the end of the day tomorrow! So.... fingers crossed!!

Kaesono, 2023-07-10:
Is there a file for fusion360 that I could try to work with? It seems like that may be more precise than Tinkercad for this type of work.

JonMuller, 2023-07-10:
Let me see if I can find it. I'll upload it if I can.

JonMuller, 2023-07-10:
Ok, so I found it, but it will only let me save fusion files to their cloud server. I can export, but it's just in formats like .stl or .step or .obj.

I think that's a limitation of the software for personal use. Sorry about that.

Kaesono, 2023-07-10:
That's alright, could you give dimensions though for the width of each pcb key column, etc. so I can move it out the right amount. I don't think it will be too hard for me to adapt it I just need to know how much to move it over.

JonMuller, 2023-07-10:
It's standard choc spacing, so 18.5mm should do it. The last two columns are aligned the same but the screw mounts drop down. You should be fine to use the screw mounts for the 5 col, just make sure there's a shelf on the outside to support the PCB though.

JonMuller, 2023-07-10:
Check that. shelf won't work well because of the hot swap sockets. But as long as it's supported by a shelf on the top and bottom, should be ok.

Kaesono, 2023-07-10:
Could I duplicate the screw mounts and move them 18.5mm aswell? Also what is the shelf?

Kaesono, 2023-07-10:
I was thinking of just moving the outer column out and filling in the gap by duplicating parts of it in

JonMuller, 2023-07-10:
No, screw mounts don't stay at the same line. They drop down a bit.
[image]

Kaesono, 2023-07-10:
By how much? I could just move them down, right?

JonMuller, 2023-07-10:
This is what I mean by shelf...
[image]

JonMuller, 2023-07-10:
I'm not quite sure how to get an accurate measure of how much, but the .svg files are basically just exports of the cut files on kicad PCB, so if you line everything up with that, you should be good to go.

Kaesono, 2023-07-10:
What would the problem be with say moving the outer 10mm or so and filling in the remaining ledge?

JonMuller, 2023-07-11:
10mm wouldn't be enough. Needs to be at least 18.46. But otherwise, shouldn't be a huge problem.

Kaesono, 2023-07-11:
Are the two halves mirrored and otherwise identical to each other?

JonMuller, 2023-07-11:
Yup. Exact mirror. The benefit of a reversible PCB. I would just make the one change, then mirror it to make the other side. That's how I did it.

Kaesono, 2023-07-11:
Do you think the outer standoffs are necessary? I have it extended but not sure if I should put them in.
[image]

JonMuller, 2023-07-11:
No. I think just like that will be fine. The little shelf on the top and bottom should support the PCB.

Kaesono, 2023-07-11:
Okay, I should probably extend the middle bit a few tenths of a mm into the other pieces to make sure they're merged, right?

JonMuller, 2023-07-11:
If you can measure the inner walls from the left to right, should be about 138mm

JonMuller, 2023-07-11:
I thought on tinkercad there was an option to select two objects and merge them.

Kaesono, 2023-07-11:
There is but I want to make sure that it prints in one piece instead of three pieces that happen to be directly next to each other.

Kaesono, 2023-07-11:
Also I measured the inside wall-to-wall as ~138.5

JonMuller, 2023-07-11:
Depending on what you're printing on, you may want a little breathing room. I've only printed on a resin printer and things shrink a little. FDM might be better about not shrinking.

Kaesono, 2023-07-11:
Hmm I will consider that. I'm gonna try now on my Prusa Mk3 in PETG and see how it goes.

JonMuller, 2023-07-11:
I do not glue the battery in at all, but I don't see why you couldn't. I think the resin print shrinks just enough so it's pretty snug, but I did put in some screws just to be sure. Screws in PETG probably work better than in resin.

Kaesono, 2023-07-11:
I ended up printing the reft side in PLA and still have to do the right side. What kind of screws/glue would you use for the keyboard?

Kaesono, 2023-07-11:
Also what feet do you use on the bottom of your case? Mine are too small for the holes.

JonMuller, 2023-07-11:
I used M2 screws pretty much everywhere, but I don't use glue, so not sure what would be best. With no glue, the battery stayed in place pretty well.

I use 8mm bumpons. I've used some by 3M and some generics. Both fit just fine.

Kaesono, 2023-07-11:
Alright,I’ll buy some 8mm ones. Which screws do you use? Mine are too small and don’t hold the pcb down.

JonMuller, 2023-07-11:
I just bought a pack of variable length M2 screws and it came with washers, so I have those and I think it's a 4mm length M2 screw.

Like this:
https://www.amazon.com/dp/B0BB1FJJKC/ref=redir_mobile_desktop?_encoding=UTF8&aaxitk=cbac584fb9b5607961ff239981298ace&content-id=amzn1.sym.cd95889f-432f-43a7-8ec8-833616493f4a%3Aamzn1.sym.cd95889f-432f-43a7-8ec8-833616493f4a&hsa_cr_id=0&pd_rd_plhdr=t&pd_rd_r=4f1f7d0f-2c45-4454-ae98-66715233a6f3&pd_rd_w=L0WCP&pd_rd_wg=9ajOw&qid=1689112354&ref_=sbx_be_s_sparkle_mcd_asin_1_title&sr=1-2-9e67e56a-6f64-441f-a281-df67fc737124

Kaesono, 2023-07-12:
I bought the pack of screws and washers and will try to put those in without the plastic breaking as it did when I tried to put my own M2 screws in recently. I also uploaded my [model to printables](https://www.printables.com/model/525505-choc-spaced-corne-xiao-6-column-case) in case anyone might need it. Are you sure that the screws and washers are big enough to hold the board in place but small enough to not interfere with the switches?

JonnyHaystack, 2023-07-28:
I have modelled this from scratch in build123d last week using the 6col SVG - though the SVG was a bit problematic (incorrect units, poorly formed, etc) so had to be optimised in Inkscape first. I used the screw hole locations from the SVG, so the 6th column screw posts are in the right place now.

Published here: https://www.printables.com/model/538027-abominationcorne-choc-xiao-case-6-columns

I really love this PCB, and the inclusion of BOM and CPL files made this the fastest keyboard build I've ever done. Thanks a ton @JonMuller 😄

JonnyHaystack, 2023-07-28:
By the way @Kaesono I'm guessing you've probably already figured out the screw stuff by now, but I just used some pan head M2x4 self-tapping screws with standard M2 washers and it worked perfectly fine for me. Metric screws and washers are really quite standard, and I recommend using ebay if you're searching for specific screw sizes in smaller quantities and don't want to buy a big set of screws.

tunfor-job, 2024-01-07:
> I have modelled this from scratch in build123d last week using the 6col SVG - though the SVG was a bit problematic (incorrect units, poorly formed, etc) so had to be optimised in Inkscape first. I [...]

Could you share fIxed SVG?

JonnyHaystack, 2024-01-07:
@tunfor-job it's there on the Printables page, under "other files"