JonMuller/gerbers #13: [Discussion] Can I build this keyboard without a display?

s5k, 2023-09-19:
Hi Jon,

I’m interested in building a budget keyboard without a display LED. Will it function properly without one?

Please forgive my lack of knowledge (this is my first time building a keyboard). I have access to PCBs, batteries, and microcontrollers, but I’m not sure what you meant by diode directions in the readme. Could you please explain how to assemble the parts correctly?

Thank you in advance.

JonMuller, 2023-09-20:
Hi there,

You can absolutely build without a display. I find them helpful for troubleshooting in the beginning, but they aren't necessary for keyboard functionality. You don't even really have to remove the code that addresses them in the firmware. OLEDs are pretty cheap depending on where you are, but they are battery hogs. Bigger batteries help negate that. Really depends on your needs. But from a bottom line cost perspective, you can absolutely skip them.

As for the diodes, they have an anode and a cathode. If you're soldering by hand, you just want to make sure you're consistent with which side you solder the anode and cathode to. I don't mean top or bottom of the board, I mean orientation on the board. Each diode location should have a square solder hole and a round solder hole. Each diode should have a marker on one side. As long as that marker is always soldered to the same solder hole, you'll be fine. You can tweak the firmware to make it work. The point of the diode is to block the flow of electricity in one direction. So you'll either block it from the rows to the columns or the columns to the rows when you press a switch. Knowing what row/column is sending a signal is how the controller determines what key was pressed. The firmware lets you pick if the rows are the input or the output. But if you aren't careful with the diodes and some are oriented from row to column and others column to row, there's no way to address that in the firmware. They just need to be consistent. 

Hope that helps! Feel free to post any follow up questions!

s5k, 2023-09-21:
@JonMuller Thank you for responding to my question! As I can see in the build images, the left side diodes should be placed at the bottom side, and the right side should be placed at the top side, isn't it? (You can see in [this image](https://i.imgur.com/wqHnod8.jpeg)).

In the CPL file, the diode angle is 180°, which means I should place the diode in PCB from anode to cathode, right? 
[image]

P/s: If there are any unacceptable words/not formal ways to ask, please forgive me, because English is not my native language.

Thank you for your attention.

JonMuller, 2023-09-22:
Hi!
No worries on the language translation. Hopefully my English translates back appropriately for you.

As to which side the diodes belong on, the fact that some are on top and some are on the bottom is a product of the reversible design and the fact that I had the diodes put on at the factory. Since they do a single side at a time, they were all technically put on the same side, but when you flip one over to make it the opposite hand, it ends up on the bottom. If you are hand soldering, just make sure to put them all the same way, i.e. Cathode always goes to the circular hole. That's the key, keep them all the same. As long as all diodes are the same, then you can make it work in the firmware.

Hope that helps!