JonMuller/gerbers #15: Choc Spaced Corne matrix wiring

kevinstadler, 2023-10-18:
Hello,

first of all thanks for such an inspiring project! I'm trying to understand how you managed to do the matrix wiring while also supporting a display on the small number of pins on the Xiao, and it's quite difficult to figure this out from the Gerbers alone.

I know that some other Xiao keyboards (like the Zaphod lite) make use of an extra shift register to increase the matrix size, but your design seems to use only the Xiao pins. But with 3 GPIOs going to the display that only leaves you 8 GPIOs for the matrix, which makes me wonder how you managed to get 18 keys (or even 21 on the 6 column layout) into the matrix? Would it be possible for you to share a screenshot of the schematic maybe?

Many thanks!
Kevin

JonMuller, 2023-10-18:
Hi there!

It's actually not that complicated. If you look at the pics in the link, the last couple pictures show you a shot of the underside. The XIAO has the 11 easily available side pins, but it also has 2 semi accessible pins on the bottom that can be used. They are originally the NFC pins, but you can reset them to be just GPIO. That gives you 13 pins to work with instead of 11, so no shift register or any other fancy stuff required, just some slightly tricky soldering. The display does need to high speed pins, so I use the regular SDA/SCL pins for that, but then I borrow one of the NFC pins for the CS. Then I use the other NFC pin for the outer most column. So technically if you only cared about the 5 col version with no memory in pixel display (you could still use the older OLED displays as they only require the SDA/SCL pins), you wouldn't need to worry about the NFC pins.

Using the NFC pins isn't everyone favorite option  because they aren't as easily accessed. If you really want to get more GPIO, there's technically 3 more pins up top where the sense module goes. If you get the non-sense version of the XIAO and you have a steady hand, you can have a total of 16 GPIO on the XIAO. Those ones are pretty tricky though and I only used them in one design just to see if I could. 

Does that answer your question? I can show you the schematic, but I don't know that they have an actual footprint for the NFC pin version. I had to make mine custom. And it took a little tweaking to get the positioning right. It also means basically permanent surface mount for the controller. I may eventually get a hot plate or something so I can try to desolder the controllers I have on boards where I messed up the routing, but not sure spending a couple hundred dollars to save a $10 controller is worth it just yet...

JonMuller, 2023-10-18:
[image]

This is the basic premise.

kevinstadler, 2023-10-18:
This is amazing, thank you so much!! My first Xiao's are still in the mail, so I wasn't aware of all the extra pins that can be used. I'd really love to include an encoder as well, so I might have to have a go at those Sense pins... I guess there's no other way than running some jumper wires to the top of the chip?

Love the reversible design with the jumpers as well! I saw in a more recent post you mentioned playing with a new reversible Xiao footprint, did you have any luck with that?

Best!

JonMuller, 2023-10-18:
That is exactly what I did with the extra sense pins. I added some through hole connectors right by where the sense pins would be once the controller was attached and I just used some magnet wire to jump from the controller to the PCB. I was able to get the encoders working without too much trouble. Encoders are better off with high speed pins, which fortunately 2 of the top 3 pins are. I've heard others get nervous about using pins so close to the antenna, but I never noticed any Bluetooth issues. 

Yes, the reversible footprint worked out just fine. It's much harder to line up the controller though. Well, not harder, just takes more patience and concentration. There's no through holes to help you line it up, but also you don't have to deal with any jumpers for the regular pins. I still have jumpers for the display, but that's it. Makes routing a lot easier. I have a working chocofi with that setup that I probably should add all the gerbers for.

kevinstadler, 2023-10-22:
It looks like Xiao footprints are now also available in the https://github.com/crides/kleeb Kicad symbols library, I guess I'll have to give these a shot. Many thanks again for your help!

closed 2023-10-22 by kevinstadler