JonMuller/gerbers #17: Castellations holes are required for JLPCB?

bullhugger, 2024-06-01:
> You need to pay additional 41.1 for your order. GERBER-corne-choc-xiao_Y6【4699775A_Y6】
> 
> Reason:
> 
> 1.Castellations are plated through holes located in the edges of a printed circuit board and cut through to form a series of half holes,it will take more time to make it ,so you may need to pay for an extra cost for it.

The cost is twice the normal manufacturing if including castellation holes. Is it required? Otherwise this is going far too expensive for me.

JonMuller, 2024-06-01:
Seems as though you've got an ambitious reviewer of your board. I have never paid for that and I haven't had any issues yet, but I do know what they are talking about. I've had to clean up the cuts a little before assembly. 

If they will let you do it without paying the extra, just say you are fine with consequences. If they won't, there isn't much else you can do. 

As to the actual consequences, the way the traces get to the pads on those holes when they are cut shouldn't be at all impacted by the cuts so long as the pads stay on the top and bottom of those holes. Worst case scenario, you'd need to bodge a wire from the switch to the battery pad on the controller. But if you're ordering 5 pcbs, the likelihood that 4 out of the 5 wouldn't work seems really low to me. Especially since it's never happened on any of mine.

bullhugger, 2024-06-02:
> Seems as though you've got an ambitious reviewer of your board. I have never paid for that and I haven't had any issues yet, but I do know what they are talking about. I've had to clean up the cuts [...]

Thanks for the info. Yeah I'm gonna proceed without the castellations hole. The reviewer was indeed very ambitious, and I don't think its worth twice of the pcb price. Thanks again.

bullhugger, 2024-06-03:
Requested JLCPCB to proceed without the castellations hole

closed 2024-06-04 by bullhugger