JonMuller/gerbers #18: Raw KiCad files?

lazerwalker, 2024-06-16:
Hi!

I'm curious if you'd consider making the full KiCad project, instead of just the gerbers, open source — your board checks off all my boxes except for not supporting replacing a key or two with rotary encoders or roller encoders, and I'd love to just fork and make that little tweak myself rather than have to recreate a lot of the layout you've already done from scratch or starting from a different project that's farther off from my goals.

Totally understand if that's not something you're willing to do!

JonMuller, 2024-06-24:
If you're up for an experiment, I can try just doing the bare minimum of effort and upload the files as is. You can then grab them and see what's broken?

lad1337, 2024-11-03:
Hi I‘m working on a keyboard that uses the xiao and want it to be reversible.
I believe you made a reversible footprint for the xiao based on your post on Reddit 
https://www.reddit.com/r/ErgoMechKeyboards/s/LwAzCU8178

As I’m still new to KiCad it would help me a lot to understand if you could upload it.

JonMuller, 2024-11-04:
You are correct! I did make a reversible XIAO footprint. I have only used it once, but it did work. The trickiest thing about footprints from what I remember is that you just need to assign the same pad number to the right place. So when you're working with the top side, say you assign the upper left pad the number one, then you need to also assign the number one to the bottom pad on the upper right. And you can't use through hole options when doing that. That just ensures that when you route things, they are going to the correct place. I'll see if I can figure out how to extract a footprint. I'm a bit busy right now, so it may take some time.

I will also say from experience that just looking at a footprint in KiCad was not super helpful to me. Not everything was obvious by sight. I had to really dig into the settings/config of it to understand what all was happening.

lad1337, 2024-11-05:
hey thanks for your feedback.
i gave it a shot and created a shot and created https://gist.github.com/lad1337/6de9173f0fbd4676588e0da8372f2eef have not yet tried it

JonMuller, 2024-11-08:
No worries @gabriel-viviani. That 3rd orange keyboard is a corn-ish zen. It uses an e-ink display. Depending on how good you are (if this is your first build, then I would say maybe not ready yet, but maybe just first keyboard build? not first soldering experience?) you could possibly add one of these, but they require at least 5 gpio as opposed to the 3 for the N!V or 2 for the OLED. They are a bit more expensive than the OLEDs though, but still cheaper than N!V. If I'm being completely honest though, I do not care for them. The refresh rate is pretty tough to get used to. Most people just use them for overall connection status and battery levels, not for layer indicators or anything else. Your most cost effective option would be to go with an OLED and then look for a battery that is in that 500mAh+ range. I really like the Nintendo JoyCon batteries for this because I think they're likely the easiest to source? but I could be wrong on that. 

I hope it goes well and you enjoy it! Please feel free to ask any other questions. You can ask here or try bugging me on reddit.