JonMuller/gerbers #20: Diode and Battery connector

johnnytshi, 2024-12-06:
Hi,

My PCB has finally arrived, and I’d like to double-check two things before I start:

1. For the diodes, in your photo I see that on the right-hand side they are soldered on the top rather than the bottom (due to PCBA constraints). My understanding is that I can place them on the bottom side, as long as the cathode aligns with the circular hole. Is that correct?

2. For the battery connector, I believe I need [[this component](https://www.aliexpress.com/item/1005006478369838.html?spm=a2g0o.productlist.main.5.b15777dfOm14Z2&algo_pvid=0aca5eea-7ec6-410a-bab8-cfa60e97509e&algo_exp_id=0aca5eea-7ec6-410a-bab8-cfa60e97509e-2&pdp_npi=4%40dis%21EUR%211.04%210.91%21%21%211.07%210.94%21%402103146f17335228783672172ea2d1%2112000037348813004%21sea%21IE%210%21ABX&curPageLogUid=gaTTq5Iqr138&utparam-url=scene%3Asearch%7Cquery_from%3A)](https://www.aliexpress.com/item/1005006478369838.html?spm=a2g0o.productlist.main.5.b15777dfOm14Z2&algo_pvid=0aca5eea-7ec6-410a-bab8-cfa60e97509e&algo_exp_id=0aca5eea-7ec6-410a-bab8-cfa60e97509e-2&pdp_npi=4%40dis%21EUR%211.04%210.91%21%21%211.07%210.94%21%402103146f17335228783672172ea2d1%2112000037348813004%21sea%21IE%210%21ABX&curPageLogUid=gaTTq5Iqr138&utparam-url=scene%3Asearch%7Cquery_from%3A). Should I solder all four points? are the additional two just for extra structural support?

All the best,

Johnny

johnnytshi, 2024-12-06:
forgot to ask, what is J9 for? doesn't seem like thats used based on your pictures

JonMuller, 2024-12-06:
Hi,

1. Correct. Just keep everything facing the same way and worst case scenario you just have to adjust the firmware for diode direction. 

2. That does look like it should work. It's hard to say for sure without actual physical measurements, but if it's designed to work with a joycon connector, then it should fit the footprint on the board. The additional points on the back of the connector are both connected and for structural support. They are connected to GND, but if you don't solder them, it will still work. But if you unplug the battery, do so very carefully. It's generally a tight fit to begin with.

As for J9, that is just a secondary spot to solder a battery without a connector. So if you were using any number of available lipos, you could just solder them directly and not worry about the connector. 

Hope that helps!

johnnytshi, 2024-12-07:
Make perfect sense. Thanks again!

closed 2024-12-07 by johnnytshi