JonMuller/gerbers #22: Having trouble build row2col for corne-choc-xiao

johnnytshi, 2025-01-17:
Hi,

Finally got my Xiao after the long Canada Post delay. Do you know what I need to change to build row2col?

Here is what I did:

1. moved all the files inside zmk_starter/view into a zmk-config (added a build.xml)
2. inside chipper.dtsi, change the diode-direction = "row2col";

it did build, https://github.com/johnnytshi/zmk-config/tree/master
however, when i press a key, the keyboard would issue seemingly random keys, for example, when i use evtest, and press "a", i get the following:
Testing ... (interrupt to exit)
Event: time 1737144001.653905, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70014
Event: time 1737144001.653905, type 1 (EV_KEY), code 16 (KEY_Q), value 1
Event: time 1737144001.653905, -------------- SYN_REPORT ------------
qEvent: time 1737144001.869496, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70083
Event: time 1737144001.869496, type 1 (EV_KEY), code 240 (KEY_UNKNOWN), value 1
Event: time 1737144001.869496, -------------- SYN_REPORT ------------
Event: time 1737144002.121851, type 1 (EV_KEY), code 240 (KEY_UNKNOWN), value 2
Event: time 1737144002.121851, -------------- SYN_REPORT ------------
Event: time 1737144002.155421, type 1 (EV_KEY), code 240 (KEY_UNKNOWN), value 2

and it does NOT end

I also retried the firmware here: https://github.com/JonMuller/gerbers/tree/main/corne-choc-xiao/zmk_starter/view/firmware/row2col
The left one works, (however, i need 6 col instead 5). But the right one did NOT work at all.

any idea what I should do? 

All the best

Johnny

JonMuller, 2025-01-17:
First, let me apologize. I did not consider 6 column people at all when I added all the starter code and I regret that. It probably makes it look easier than it is. 

So for starters, you'll want to change the transform to the 6 column default one. On line 12 in your dtsi file:
Change = five_column_transform
to = default_transform

Then you'll need to add &kp [insert whatever key you want] at the beginning and end of the top 3 rows for each layer in the keymap. Thumbs are find because nothing changes there. So if that's &kp ESC or &kp TAB or whatever, you'll need to add those in so those outer keys actually work. I think you'll get an error on build if you don't at least put &trans in there. 

Also, all those combos will be off from their intended location because now the count starts one switch to the left. 

Once you make those changes and you successfully compile, then load that and see where we're at. 

To be fair though, most of that shouldn't matter. When you loaded the test firmware and the left side worked (minus the outer column) that suggests the board and controller are sound, so your firmware should have worked.

How much of the assembly did you do yourself? That would be the next thing to look at. All the soldering and such.

johnnytshi, 2025-01-18:
I did all the assembly by hand. I built the Sofle Choc that I am using currently, but I have been slowly removing more keys , now to the point I can move to a corne style wireless keyboard, and this is probably the slimmest one.

ok, good news is finally all the keys worked. i think "CONFIG_NFCT_PINS_AS_GPIOS=y" tripped me up. now the remaining issue is my right side. The firmware still doesn't respond to any key presses. 

Interestingly, if I flash the left-side firmware onto the right side, all six columns work correctly—though, of course, the key mapping is incorrect. The same is true for the firmware provided by you, right side would respond to keys if flashed with lift side firmware

any idea?

closed 2025-01-18 by johnnytshi

JonMuller, 2025-01-19:
Sorry! Didn't see this until the morning. No worries at all. I would just remove the stuff about you being dumb. There's often stuff I take for granted being around this for awhile now. It's good that you documented where your exact issue was. You are correct. The right is just a peripheral. It needs the left to work completely. 

I hope it works well for you and you enjoy it!

johnnytshi, 2025-01-20:
It works great, now I am just waiting for the battery + connector

For anyone else interested, this is a really great keyboard. It is a bit narrower than corne (I only have v4 to compare), which I find somehow makes it more pleasant to type. Also, the thumb cluster is more ergo compared to corne